Rohanvamshi/Dark_bulb #16: Add Project Directory Outlines

bama4, 2020-08-11:
**Description**
This issue involves adding a bare project structure to the master branch
The structure of the project is as follows:
```
dev_gpio
                       |_____________   Makefile
                       |_____________   src
                       |_____________   include
bulb
                       |_____________   Makefile 
                       |_____________   src
                                              |_____________ bulb.c
                                              |_____________ bulb.h  
                       |_____________   include
Makefile
scripts
                       |_____________   install_devgpio.sh
                       |_____________   install_bulb.sh
man
                       |_____________   bulb.man
```

**Prerequisites**
Design documents are done

**Goal**
This ticket requires the following:
- [ ] Create stubbed out project

**Definition of Done**
This ticket is completed when this is merged into master

bama4, 2020-08-12:
@Rohanvamshi I believe that the include header files that you put in:

```
    |_____________   include
                     |_____________ linux/kernel.h
                     |_____________ linux/module.h
                     |_____________ sys/ioctl.h
```
Are system header files and we won't have to explicitly place them in our project directory

Rohanvamshi, 2020-08-12:
@bama4 Dint know they were system headers. Missed that point. Made the changes. Check and revert if any more changes need to be done.

closed 2020-08-17 by Rohanvamshi