Rohanvamshi/Dark_bulb #17: Add initialising module function to GPIO_Device_Driver.c

Rohanvamshi, 2020-08-12:
**Description**
This issue involves creating the function in GPIO_Device_Driver.c for initializing the kernel module

**Goal**
This ticket requires the following:
- [ ] Create a function to print an alert message to the kernel system log when GPIO_Device_Driver.ko (kernel object)  is added to the kernel

**Prerequisites**
None

**Definition of Done**
This ticket is completed when the function is implemented in GPIO_Device_Driver.c and tested

bama4, 2020-08-12:
You should be able to print the initialization message by calling `printk`  in the function that is passed to `module_init` in your kernel module source code.
This ticket may be better off being one of several steps in a single ticket to write the `module_init` function for the kernel module

Rohanvamshi, 2020-08-12:
Tittle seems to be a little misleading. I wanted it to be a ticket for module_init as well. Will make the changes soon 👍

closed 2020-08-23 by Rohanvamshi