Rohanvamshi/Dark_bulb #18: Add function to print cleanup confirmation message for GPIO_Device_Driver.c

Rohanvamshi, 2020-08-12:
**Description**
This issue involves creating the function in GPIO_Device_Driver.c for printing a cleanup confirmation message when the kernel module is removed from the kernel space. 

**Goal**
This ticket requires the following:
- [ ] Create a function to print an alert message to the kernel system log when GPIO_Device_Driver.ko (kernel object)  is removed from the kernel

**Prerequisites**
None

**Definition of Done**
This ticket is completed when the function is implemented in GPIO_Device_Driver.c and tested

bama4, 2020-08-12:
As I mentioned in #17 , since we can print messages using `printk` this may be better lumped with the ticket that implements the cleanup function passed to `module_exit`

bama4, 2020-08-21:
Had to do module init and exit at the same time to test that the device properly was created and removed

closed 2020-08-23 by Rohanvamshi