Rohanvamshi/Dark_bulb #23: Add function to read from GPIO pin in GPIO_Device_Driver.c

Rohanvamshi, 2020-08-12:
**Description**
This issue involves creating the function in GPIO_Device_Driver.c that reads specified GPIO pin and returns the state to bulb.c 

**Prerequisites**
GPIO_Device_Driver.c user interface is defined.
The state diagram describing interactions between hardware and GPIO_Device_Driver is defined

**Goal**
This ticket requires the following:
- [ ] Create a function(s) to read specified pin and returns the state to bulb.c

**Definition of Done**
This ticket is completed when the function is implemented in GPIO_Device_Driver.c and tested

bama4, 2020-08-26:
This ticket essentially implements the ioctl "read" function.
Typically  the `copy_to_user` function is used to transfer data from kernel memory to user memory

bama4, 2020-09-04:
@Rohanvamshi you may be interested in this:

I wrote a program that reads from /dev/mem to test the memory address.
The program prints out the first 4 bytes starting from the location of the GPIO registers for reading the state of the gpio pins.
if you are changing a pin between 0-31, you will see the corresponding bit position change.

I thought this may help for debugging other register functions:
```
#include<stdio.h>
#include<stdlib.h>
#include<sys/mman.h>
#include<sys/types.h>
#include<sys/stat.h>
#include<unistd.h>
#include<string.h>
#include<fcntl.h>
#include<stdint.h>

#define GPIO_BASE 0x3f200000
#define GPIO_0_32_OFF 0x34
#define MAP_ADDR 0x00000000
#define MAP_SIZE 0xb3

void int_to_bin(uint32_t num){
  for (uint32_t ctr = 1 << 31; ctr > 0; ctr = ctr / 2){
    (num & ctr)? printf("1"): printf("0");
  }
  printf("\n");
}

int main(int argc, char ** argv){

  int devmem_fd = open("/dev/mem", O_RDONLY);

  if(devmem_fd < 0){
      printf("Error opening file\n");
      goto leave;
  }

  void *buffer = mmap((void *)MAP_ADDR, MAP_SIZE, PROT_READ,
    MAP_SHARED, devmem_fd, GPIO_BASE);

  if(buffer < 0){
    printf("Error on mmap\n");
    goto leave;
  }

  uint32_t *gpio_val = (uint32_t *)(buffer+GPIO_0_32_OFF);
  printf("GPIO 0 - 32\n");
  int_to_bin(gpio_val[0]);
  int_to_bin(gpio_val[1]);
  int_to_bin(gpio_val[2]);
  int_to_bin(gpio_val[3]);

  //deallocate virtual memory
  munmap(buffer, MAP_SIZE);

  close(devmem_fd);

  leave:
    return 0;
}

```

Rohanvamshi, 2020-09-04:
This is great, I was writing one myself using the map function and /dev/mem to debug the code I wrote so far.

bama4, 2020-09-04:
> This is great, I was writing one myself using the map function and /dev/mem to debug the code I wrote so far. Did you add this in the read_ioctl branch?

@Rohanvamshi I didn't add it to the read branch since its moreso "debug" code rather than "release". What do you think?

Rohanvamshi, 2020-09-04:
I was thinking we can include a debug directory and include the code in it. It's an important part to test the code. Then in the final version, we can remove the debug directory if we want to and simply post the final code. What do you think about that?

bama4, 2020-09-10:
> I was thinking we can include a debug directory and include the code in it. It's an important part to test the code. Then in the final version, we can remove the debug directory if we want to and simply post the final code. What do you think about that?

@Rohanvamshi Yes, this works. We can include the debug code in our repository. We can always control the compile settings (have a TEST MODE option and RELEASE MODE option for compilation) to separate debug and release code.

bama4, 2020-09-10:
@Rohanvamshi After testing the driver a few days ago, I noticed that we don't have the `.unlocked_ctl` function defined in the `struct file_operations` structure for the driver. After I added that, the SEGFAULT error was gone.

Another issue that I encountered is that because there is already an existing gpio driver, a SEGFAULT occurs when requesting usage of the gpio address space through `request_mem_region`. Even though `request_mem_region` is technically deprecated, it is best practice in our case to use `request_mem_region`...but it appears that for our purposes we may want to not check the mem region to avoid making this first project too complicated.

Rohanvamshi, 2020-09-10:
> @Rohanvamshi After testing the driver a few days ago, I noticed that we don't have the `.unlocked_ctl` function defined in the `struct file_operations` structure for the driver. After I added that, the SEGFAULT error was gone.
> 
did you mean unlocked_ioctl? 
> Another issue that I encountered is that because there is already an existing gpio driver, a SEGFAULT occurs when requesting usage of the gpio address space through `request_mem_region`. Even though [...]

Yes, I read about request_mem_region but It seems to be a standard rather than it doing something useful in the code. If its a best practice and still being used to maintain the standard then we can include it. It looks relatively simple to implement.

bama4, 2020-09-11:
@Rohanvamshi if we include request_mem_region, our driver, we will have to disable the existing gpio driver which may add onto the complexity of this project. Maybe for now we don't use request_mem_region and try to disable built in gpio later? The main purpose of the function is to prevent two drivers from requesting virtual memory from the same physical address space. Its simple to add this function but not sure how complex it is to disable the built in GPIO driver, we would probably have to search and modify the .dtb device tree files to disable it

closed 2020-09-22 by bama4